MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #8: WiFiClient does not support https

pc0808f, 2017-05-02:
WiFiClient does not support https

pablosun, 2017-05-02:
Good feature suggestion.

@pc0808f Do you have any suggestions on desired interface? e.g.

1) Extending the `WiFiClient` class, or Create a new class `WiFiSSLClient`, similar to the design of Arduino 101? Note that adding SSL/TLS support for HTTPS require more flash size (for certificates) and RAM size. So probably it might be better to separate into 2 classes.

2) Should the library asks users to provide Root Certificate? It will save some space if we ask user to provide the exact Root CA that they would need to connect to the remote server. Otherwise we'll need to preload several certificate for different websites.

pc0808f, 2017-05-02:
The SDK support HTTPS,

Is it possible to use it?

pablosun, 2017-05-02:
Yes, but it will need some modifications to the Add-on package. 

The SDK dependency looks like this:

 * `httpclient` depends on
 * `mbedTLS` depends on
 * `lwIP` depends on
 * FreeRTOS and MT7697 driver

Currently, the Arduino Add-on only includes `lwIP` but not `mbedTLS` and `httpclient`.

So I think we can rephrase this issue to:
@pc0808f Do you need a general SSL/TLS connection API (exposing mbedTLS library) or an HTTPS client (exposing `httpclient` library)?

pc0808f, 2017-05-02:
if it can support general SSL/TLS connection API (exposing mbedTLS library), it will be better. because, i may use the websockets.

pablosun, 2017-05-09:
Hi @pc0808f Would you mind give the alpha version a try?

* Replace your board manager url to http://download.labs.mediatek.com/package_mtk_linkit_7697_dev_index.json
* Install the version **v0.7.5-alpha**
* Open the example **LWiFi > TLSClient** and modify the WiFi SSID/password. 
* Open Serial Monitor, set baudrate to **115200**

The expected result should be something like this:
```txt
Attempting to connect to SSID: MediaTek_Labs
Connected to wifi
SSID: MediaTek_Labs
IP Address: 192.168.1.49
signal strength (RSSI):-61 dBm

Starting connection to server...
connected to server (GET)
HTTP/1.1 200 OK
Content-Length: 662
Access-Control-Allow-Origin: *
Connection: close
Content-Type: application/json
Date: Tue, 09 May 2017 10:15:26 GMT
Strict-Transport-Security: max-age=631138519; includeSubdomains; preload

{"given_cipher_suites":["TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_3DES_EDE_CBC_SHA","TLS_PSK_WITH_AES_256_CBC_SHA384","TLS_PSK_WITH_AES_256_CBC_SHA","TLS_PSK_WITH_AES_128_CBC_SHA256","TLS_PSK_WITH_AES_128_CBC_SHA","TLS_PSK_WITH_3DES_EDE_CBC_SHA","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"],"ephemeral_keys_supported":false,"session_ticket_supported":false,"tls_compression_supported":false,"unknown_cipher_suite_supported":false,"beast_vuln":false,"able_to_detect_n_minus_one_splitting":false,"insecure_cipher_suites":{},"tls_version":"TLS 1.2","rating":"Improvable"}
disconnecting from server.
```

Please kindly comment if this suits your needs, and feel free to feedback bugs or request in this thread.

thanks!

pc0808f, 2017-05-10:
i will try it , and get so feedback later~

pc0808f, 2017-05-11:
The TLSClient  maybe work, but i only can receive 32768 bytes data from client.read 

        char c = client.read();
        Serial.write(c);
        count++;

I use a count to count read data.

My usage is to download to FOTA bin, but it seem can not receive a big bin file(183,336bytes).

BTW, the both the TLSClient  and WIFIClient can not link to http(80) anymore??

the following result is print from example wifiwebclient

Attempting to connect to SSID: sam
Connected to wifi
SSID: sam
IP Address: 10.0.2.11
signal strength (RSSI):-31 dBm

Starting connection to server...
connected to server (GET)

disconnecting from server.

pablosun, 2017-05-11:
1) The wrapper class itself does not pose such constraint - but the number "32768" looks like a bug somewhere... I'll reproduce this problem.

2) TLSClient always performs a TLS handshake after connection, so it cannot connect to "normal" HTTP port, as the handshake will fail. However, there are no modifications to the `WiFiClient` class in this alpha version. So this is a separate issue that we need to look into. 

I can run the `WiFiWebClient` example and retrieve the text file successfully. To reproduce this problem, can you please help the following experiments?

  * Share the host/port address that you use
  * Run the `WiFiWebClient` example. If it does not work, try modify the last `delay` statement in the `setup()` function - it looks suspicious to me.
```c
    if (client.connect(server, 80)) {
        Serial.println("connected to server (GET)");
        // Make a HTTP request:
        client.println("GET /linkit_7697_ascii.txt HTTP/1.0");
        client.println("Host: download.labs.mediatek.com");
        client.println("Accept: */*");
        client.println("Connection: close");
        client.println();
        delay(500);          // modify to a longer delay, e.g. 500ms.
    }
```

pc0808f, 2017-05-11:
It is better if the delay set to 500.

So, let we back to count number.

both wificlient and TLSclient can not get whole files.

you can try this

    if (client.connect(server, 443)) {
        Serial.println("connected to server (GET)");
        // Make a HTTP request over SSL (HTTPS)
        client.println("GET /pc0808f/MySonCar/master/linkit.bin HTTP/1.0");
        client.println("Host: raw.githubusercontent.com");
        client.println("Accept: */*");
        client.println("Connection: close");
        client.println();
        delay(500);
    }

and the counting code

    while (client.available()) {
        char c = client.read();
        count++;
        Serial.write(c);
    }

pablosun, 2017-05-11:
Thanks, I can reproduce the issue. Looks like there is an `0x20000538: bad message length` error from mbedTLS library after reaching byte 32768 - will keep looking into it.

pablosun, 2017-05-17:
Update issue status:
 * This problem can be reproduced with the LinkIt SDK's `mbedTLS` example project
 * Adjust the `MBEDTLS_SSL_MAX_CONTENT_LEN` from 6KB to 16KB `(16*1024)` solves this problem
 * However, doing so on the **Arduino BSP** results in `X509 Verification Error`. Looks like memory corruption issue.

We'll need to keep investigate into the `X509 Verification Error` issue before we can apply the change to  `MBEDTLS_SSL_MAX_CONTENT_LEN`.

pablosun, 2017-05-17:
Turns out the verification error is caused the insufficient heap size. The Arduino BSP consumes more heap memory then its LinkIt SDK projects (because of the wrapping C++ objects). In the mbedTLS library, the `bignum.c` module dynamically allocates small chunk of heap memory in functions like `mbedtls_mpi_exp_mod`. If the allocation fails, the verification fails.

So, the **stop at byte 32768** issue can be fixed by:
 * Increase `MBEDTLS_SSL_MAX_CONTENT_LEN` to `16KB`, which follows the RFC standard
 * Increase `configTOTAL_HEAP_SIZE` in `FreeRTOSConfig.h` in Arduino BSP layer.

pablosun, 2017-05-18:
Hi @pc0808f,

 * Upgrading to `0.7.5-alpha2` from http://download.labs.mediatek.com/package_mtk_linkit_7697_dev_index.json should fix the **32768 byte** issue.
 * You might need to manually remove `0.7.5-alpha` folder from the directory `%USERPROFILE%\AppData\Local\Arduino15\packages\LinkIt\hardware\linkit_rtos\` after upgrading the package.

Hope this helps :)

pablosun, 2017-05-23:
Supported in v0.7.10.

PR #14 (2017-05-19): Fix #8: Support SSL/TLS connection by adding TLSClient
We created a TLSClient class that wraps mBedTLS.

 * mbedTLS is already in the `liblinkit.a`
 * Increase FreeRTOS heap size to 120KB
 * Switch the default mbedTLS config file to `config-mtk-websocket.h` which have a larger TLS content buffer.
 * Add example project **TLSClient.ino**
  * Also fixes #12 by adjust the input ssid string to `const char*`
closed 2017-05-23 by pablosun