MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #11: [Feature request] Notification support of LBLECharacteristic

SmallpTsai, 2017-05-14:
Currently it seems LBLECharacteristic does not support notification
Notification allow central side receive notification of value change.
Ex. 7697 side (as peripheral) simply to update weather informations and Mobile side (as central) to receive notification and process.

SmallpTsai, 2017-05-14:
FYI, CCCD (Client Characteristic Configuration Descriptor), 0x2902, 0x0001

https://devzone.nordicsemi.com/tutorials/17/

Notice also that "Notify" is added to the Characteristic Declaration's properties field. The UUID value of the CCCD is 0x2902 as we have seen earlier and the value is 0x0000 meaning that notification and indication is currently switched off. If you want you can enable notification right away, but since we have not yet set up any mechanisms updating the values nothing fancy will happen. Anyway, try to click the "Enable services" button. You should see that the value field of the CCCD is updated to 0x0001, meaning that notification is indeed enabled. You can also use the "Write" button and write to the CCCD manually.

pablosun, 2017-05-15:
Good feature suggestion, I'll add this to the enhancement backlog.

pablosun, 2017-08-21:
Will be available in v0.9.2

closed 2017-08-21 by pablosun