MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #13: [MCS Library] socket.connect() failed after uploading data points for several times.

dariachen, 2017-05-16:
Upload integer, hex and string data points 13 times for each and got socket.connect() failed, socket.status()=0 in "bool MCSDevice::_uploadDataPoint(const String& params)" method.

pablosun, 2017-05-18:
There are memory leaks. After several TCP connection being created and closed, the follow error log is emitted:

```
[Arduino:atcp_start:716] TTCP [0x2000b4b8]: could not allocate payload
```

By enabling the option `MTK_SUPPORT_HEAP_DEBUG = y` in `feature.mk` we are able to trace the leakage is caused by the `atcp_start` function:
```c
	if (ttcp->mode == TTCP_MODE_TRANSMIT) {		//for client

		int8_t id = insertNewClientConn(ttcp, p);	//always insert 0.
		if (id == NO_VALID_ID)
			return ERR_MEM;

		//Global val, Indicate client whether connect server success.
		_connected[ttcp->sock] = 0;

		ttcp->payload[id] = malloc(ttcp->buflen);	//malloc receive buf
```

It seems that `ttcp` is not mananged properly. Need further investigation to find where the leakage happens.

pablosun, 2017-05-19:
should be fixed by version **v0.7.6-alpha**, which is available in the development board manager URL:
```txt
http://download.labs.mediatek.com/package_mtk_linkit_7697_dev_index.json
```

commit 3f082040 (2017-05-23): Re-implement WiFi socket features (#16)

Current implementation is based on asynchronous LwIP APIs, and there are memory management issues that cause #13.

We re-implmemented the sockets with LwIP BSD-style socket APIs, which are blocking calls that are better suited to Arduino calling styles and easier memory management. This fixed #13.

This pull request also have following improveent:

 * Reduced waiting time for WiFi.begin().
 * Remove redundant checks and waits in WiFi example sketches. (The 10-second wait after `begin()` is unnecessary).
 * Adds MCS classes' syntax coloring to fix issue  [... cut]
closed 2017-05-23 by pablosun