MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #20: Arduino IDE 1.8.2 : "Aggressively Cache Compiled Code" will cause compile error.

bearwbearw, 2017-05-25:
[image]

When enabled the "aggressively cache compiled code" (new feature of 1.8.2, default is "On") in the preference menu, when open a new window by selecting a new example, will have compile error when compiling the code.

**Error Code:**
arm-none-eabi-gcc: error: C:\Users\mtk00612\AppData\Local\Temp\arduino_build_72054/core/syscalls.c.o: No such file or directory

[image]

pablosun, 2017-05-26:
This is because we explicitly links `core/syscalls.o` in our `platform.txt` file. 

The reason we need to link this file is because we need to make sure symbols required by C-Runtime won't be discarded as described in this [thread](https://sourceware.org/ml/newlib/2015/msg00670.html).

If we link the `syscalls.o` into the `core.lib` file we will get following link errors:

```log
c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-init.o): In function `__libc_init_array':

init.c:(.text.__libc_init_array+0x1c): undefined reference to `_init'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-sbrkr.o): In function `_sbrk_r':

sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-writer.o): In function `_write_r':

writer.c:(.text._write_r+0x10): undefined reference to `_write'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-closer.o): In function `_close_r':

closer.c:(.text._close_r+0xc): undefined reference to `_close'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-fstatr.o): In function `_fstat_r':

fstatr.c:(.text._fstat_r+0xe): undefined reference to `_fstat'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-isattyr.o): In function `_isatty_r':

isattyr.c:(.text._isatty_r+0xc): undefined reference to `_isatty'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-lseekr.o): In function `_lseek_r':

lseekr.c:(.text._lseek_r+0x10): undefined reference to `_lseek'

c:/users/mtk03251/appdata/local/arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc_s.a(lib_a-readr.o): In function `_read_r':

readr.c:(.text._read_r+0x10): undefined reference to `_read'

collect2.exe: error: ld returned 1 exit status
```

pablosun, 2017-05-26:
A solution is to move the `syscalls.c` file from `core` to `variant`. The `variant` won't get cached and archived into a `.a` file.

closed 2017-05-26 by pablosun

ermias1981, 2018-12-11:
Dear pablosun the same problem happens when I tried to compile my library file developed with C and Arduino IDE 1.8.5 for Teensy 3.6 hardware. while I try to compile the same problem happens as below. whould you mind to tell the solution how can I fix this issues? As you proposed above I couldn't get syscalls.c file in core .

with kind regards.

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/fpu\libc.a(lib_a-sbrkr.o): In function `_sbrk_r':

sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'

collect2.exe: error: ld returned 1 exit status

commit 1a673a2a (2017-05-26): Fix #20 by moving syscalls.c from core to variants directory