MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #28: Adding Sleep API

will127534, 2017-06-09:
Hi all,
I notice that MT7697 SDK have sleep function
so maybe Arduino API can add a sleep() function to reduce power consumption

here is the function at /core/arduino/delay.c

`void sleep(uint32_t ms){
	if(hal_sleep_manager_init() == HAL_SLEEP_MANAGER_ERROR)
    {
        // Check the log for error handling.
        return;
    }
    if(hal_sleep_manager_is_sleep_locked())
    {
        printf("Cannot enter the sleep mode, as the sleep is locked\n");
    }else
    {
        if (hal_sleep_manager_set_sleep_time(ms) == HAL_SLEEP_MANAGER_OK)
        {
            hal_sleep_manager_enter_sleep_mode(HAL_SLEEP_MODE_SLEEP);
        }
    }
}
`

pablosun, 2017-06-29:
Note that we'll keep tickless mode disabled in FreeRTOS. Therefore this API will actually put the board into WFI sleep mode - which should save more power than an "idle (delay)" sleep.

pablosun, 2017-06-30:
Hi @will127534 , how long does the `sleep` API is being able to sleep in your case?
In my case the `enter_sleep_mode` succeeded but immediately wakes up.

```cpp
  Serial.print("start sleep, time=");
  Serial.println(millis());
  bool isOK = sleep(5000);
  Serial.print("sleep result=");
  Serial.println(isOK);
  Serial.print("time=");
  Serial.println(millis());
```

And the output is always
```txt
start sleep, time=1
sleep result=1
time=5
```

pablosun, 2017-07-28:
We'll set this enhancement to pending statue first, until a more concete usage scenario appears.

closed 2017-07-28 by pablosun