MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #52: TLSClient.available() is very slow

pccr10001, 2017-08-05:
Hello,

I am using TLSClient + PubSubClient for AWS IoT MQTT Client.
And I found TLSClient.available() stop for 4 secs on mbedtls_ssl_read() function.

pablosun, 2017-09-20:
Hi @pccr10001, this seems to be server-response related.  When connecting to `howsmyssl.com` the TLSClient::available() took 1~6ms to return, which seems to be pretty fast.

If this issue persists in your case, you can try set the socket timeout option:

 1. open the file `TLSClient.cpp` in `%LOCALAPPDATA%\Arduino15\packages\LinkIt\hardware\linkit_rtos\0.9.5\libraries\LWiFi\src`

 2. find the method `TLSClient::connectImpl`

 3. After the call to `mbedtls_ssl_set_bio`, insert following code snippet:

```cpp
        int timeout_ms = 300; // set Timeout to 300ms
        struct timeval {
            long      tv_sec;
            long      tv_usec;
        };
        struct timeval interval = {timeout_ms / 1000, (timeout_ms % 1000) * 1000};
        if (interval.tv_sec < 0 || (interval.tv_sec == 0 && interval.tv_usec <= 0))
        {
            interval.tv_sec = 0;
            interval.tv_usec = 100;
        }
        setsockopt(m_cntx.net_ctx.fd, SOL_SOCKET, SO_SNDTIMEO, (char*)&interval, sizeof(struct timeval));
```

pablosun, 2018-08-15:
close until we can reproduce this issue.

closed 2018-08-15 by pablosun