MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #56: Roadmap / Documentation ?

luc-github, 2017-08-09:
Hi it looks nice project ^_^

Any plan to add OTA support library / SPIFFS or equivalent  library? or any library to bring LinkIt at same level as ESP8266/ESP32 ?

Or is there any detailled documention about SDK / memory mapping to help to port them ?

I have a project currently using ESP8266 and soon ESP32 and I would love to support LinkIt board as well :smile_cat: 

Thanks

pablosun, 2017-08-10:
Thanks for asking!

### OTA
We do have plans for OTA support in the near future. At this stage we are working on enhancing the BLE functionality.

### SPIFFS
For SPIFFS, its not yet in the roadmap but we are open to this. Do you specifically need a file system-like API that supports usage like `open( filePath, ...)`, or any persistent storage would do? If the data item is not very large (a few bytes), we would suggest using the `LFlash` class, which is essentially a key-value mapped storage.

### Documentations
For documentations, the resource pages are:
 * [MT7697 technical reference manual](https://docs.labs.mediatek.com/resource/mt7687-mt7697/en/documentation). The **MT76x7 Technical Reference Manual** contains register mappings that you might need
 * If you are interested in bootup / bootloader, please take a look at the **LinkIt SDK**, which can be downloaded at [here](https://docs.labs.mediatek.com/resource/mt7687-mt7697/en/downloads).
 * If you need other libraries to port your project, please let us know.

luc-github, 2017-08-10:
Thanks a lot for the detailed answer ^_^
Great for OTA, ^_^ as I think it is key on this kind of board
About File system : I use it to store the webserver files 
[image]
So it is more than few bytes 
the current ESP8266 libraries that I use in my project for your reference are:
- EEPROM library to store configuration data , this one is close to the flash you suggest, it has a 4Kb capacity to let the flash available for SPIFFS
- SPIFFS libraries to store webserver files and webserver configuration files, it use the space left by program, OTA, and EEPROM, so it is usually around 3Mb on 4Mb flash
- Web Server library which is an abstraction layer for GET/POST/ chunk and Upload data  
- Device anouncement libraries : NetBios, SSDP, mDns, and for Soft AP : Captive portal ( I read you plan to support Soft AP also which is nice to have to configure the device the first time 
- SD library, I used ESP8266 SD Library but now I use external one : https://github.com/greiman/SdFat which allow long file name

I like the Linkit boot loader which unlike ESP8266 do not send any garbage at start which is great for device connected to another one by serial

I will definitly read the documentations - is there a forum / gitter to avoid to poluate issue tracker when having programing questions ?

Thanks again

pablosun, 2017-08-17:
Hi, 

For discussions that are not directly related to the Arduino Board Support Package, you can visit the forum here: https://en.forum.labs.mediatek.com/c/linkit-7697

Thanks for asking!

closed 2017-08-17 by pablosun

luc-github, 2017-08-17:
Thanks a lot :smile_cat: