MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #73: Linkit 7697 in AP mode not accepting connections

kamenlitchev, 2017-11-29:
Hi, guys,

Just bought a LinkIt 7697 dev board. Trying the AP sample without modifications, I get AP to be boradcasted. However, connecting to it - Android 7 and Windows 10 (with and without static IP), does nothing.

pablosun, 2017-11-30:
Hi @kamenlitchev 

Does the serial monitor gets any output before and after connecting from the browser? e.g.
```txt
Configuring access point...AP ready.
Connect to AP LinkItAP and visit http://10.10.10.1
AP MAC=3C:15:C2:C4:8C:21
```

Now open Wi-Fi setting and connect to `LinkItAP` with password `yourpassword`, using DHCP.
After connected to the AP network, opening mobile browser and connect to `http://10.10.10.1`:
```txt
GET / HTTP/1.1
Host: 10.10.10.1
Connection: keep-alive
Upgrade-Insecure-Requests: 1
User-Agent: Mozilla/5.0 (Linux; Android 6.0.1; HTC Desire EYE Build/MMB29M) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.98 Mobile Safari/537.36
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.8

client disonnected
GET /favicon.ico HTTP/1.1
Host: 10.10.10.1
Connection: keep-alive
User-Agent: Mozilla/5.0 (Linux; Android 6.0.1; HTC Desire EYE Build/MMB29M) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.98 Mobile Safari/537.36
Accept: image/webp,image/apng,image/*,*/*;q=0.8
Referer: http://10.10.10.1/
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.8

client disonnected
```

kamenlitchev, 2017-12-01:
Hi, yes, prints out the AP settings. But nothing could connect in a stable way with WPA - Win 10 machine, Motorola G3 and iPhone 4S. I reverted to Open AP and only Windows could connect after several attempts. On the Open AP the Win machine could also issue few sporadic GET request but only under Edge (3 out of a dozen attempts), FF 47 (one out of several attempts) and no success in Chrome 62.

So, while in general it seems to be working, the working part is really matter of luck. First the connection is really spotty and Android could not even connect (in both protected and open mode). Second, even when connected, the browser request is again a matter of luck.

pablosun, 2017-12-04:
@kamenlitchev Thanks for the feedback. On my LinkIt 7697 I can connect to the SoftAP with HTC Desire Eye and an iPhone 7. 

To diagnose this problem, we need to check if this is caused by
 1. broken LinkIt 7697 HDK, e.g. bad antenna
 1. Incorrect firmware, e.g. the on-board Wi-Fi firmware is corrupted
 1. Other environment issues, e.g. a bad MAC address, or there are some incompatible Wi-Fi configuration such as regional band information.

So:
 1. Is the STA mode working normally on your LinkIt 7697? e.g. Does `Examples > LWiFi > WiFi Web Client` have a reliable connection in your Wi-Fi network?
 1. Can you choose the `Burn bootloder` option from the `Tools` menu, and check if the bootloader and Wi-Fi firmware is updated successfully? 
 1. Does the printed `AP MAC` address look normal to you? It should be something like `8C:88:2B:00:XX:YY`

kamenlitchev, 2017-12-07:
Hello,

Thanks for your continuing support! The board seems OK for me:

1. WiFi Web Client sample works flawlessly. The board connects and fetches data just fine in under 10-15 seconds total after restart (WPA protected network)
2. I updated the FW
3. Printed MAC seems fine: `8C:88:2B:00:XX:YY`

Additionally, `SimpleWebServerWiFi' works flawlessly. It connects to our WiFi and browser response times are pretty fast (under 2 seconds from pressing refresh button in browser).

With regards to the above, WiFi module of the board is operational. But softAP mode is where disaster strikes - as I said, in WPA mode no device can reliably connect with the board constantly dropping connection. Open mode is better - my Windows machine can hold the connection enough for several browser requests to go through.

kamenlitchev, 2017-12-14:
I will close it, try a non-Arduino solution and write back if not working :) Thanks, @pablosun!

closed 2017-12-14 by kamenlitchev