MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #80: Bluetooth LE notification not working

weixiongmei, 2018-01-07:
When trying to enable the notification on the Android with my own code, it failed, and the problem is related to the descriptor not found for the characteristic, the APP "nRF Connected" can't enabled the notification too.

pablosun, 2018-01-09:
Hi, thanks for reporting this issue.

Is LinkIt 7697 a Peripheral in this case? e.g. you want to receive the BLE attribute notification on the Android app nRF connected?

Is it possible to show the code snippet that you used to send the notification?

weixiongmei, 2018-01-09:
I tried all the Linkit 7697 samples that installed with the Arduino BSP, no one works, i'm sure the code that i wrote for Android is good, because i have other BLE boards such ESP32 and nRF52832, they both works with my Android code. I also tried the Linkit 7697 with the "nRF Connect" tool that's downloaded from the Android App store, but still no luck. So the only possibility is the Linkit 7697 BSP.

On the android side, when trying to enable the notification, it returns the Descriptor not found error, i don't know iOS, haven't tried that...

Not just this, there are some other bugs such as MTU not configurable(Not really a bug, but when using the LBLECharacteristicBuffer class, it will send what ever the size it is, for instance, if i don't call the setValueBuffer function, the default size of the buffer is 512 bytes,  so when i send a read request from the Android side to the Linkit 7697, i will get a 512 bytes single packet, it seems like it just ignores the MTU).

pablosun, 2018-01-09:
Hi, can you describe more clearly what the Android code does? e.g. which API retruns the "Desciptor not found" error? The example `LBLE > Notification` can be use with apps that receives BLE value notification, such as [Lightblue Explorer](https://itunes.apple.com/tw/app/lightblue-explorer/id557428110?mt=8). 

On the MTU issue, there are 2 concepts here:
 1. No, LBLE library does not provide MTU adjustment. You can achieve this by calling the LinkIt SDK API  [bt_gatts_set_max_mtu](http://labs.mediatek.com/api/mt7697/group___bluetooth_b_l_e___g_a_t_t_s.html#gac47c5fa858433972df6c6d624230139f).

 2. The characteristic value size and MTU are two different concepts - the MTU is determined by both the peripheral and the central, and the size of the value of a characteristic is defined by application developers. You can have a MTU of 50 bytes and sends a value of 4 bytes, and vice versa.

pablosun, 2018-01-09:
One possible reason is that currently the Arduino BSP does not implement the **Client Characteristic Configuration Descriptor**(CCCD), which may be used by Android API `android.bluetooth.BluetoothGatt.setCharacteristicNotification`.

weixiongmei, 2018-01-09:
bluetoothle.subscribe((e) => {
                    document.body.children[0].innerHTML += "Success:\r\n";
                    document.body.children[0].innerHTML += "[" + e.name + "] Value:" + bluetoothle.bytesToString(bluetoothle.encodedStringToBytes(e.value)) + "\r\n";
                    }, (e) => {
                    document.body.children[0].innerHTML += "Fail: " + e.message + "\r\n";
                    }, { address: e.address, service: "19B10010-E8F2-537E-4F6C-D104768A1214", characteristic: "19B10011-E8F2-537E-4F6C-D104768A1214" });

i used Cordova to write the mobile app, the above code is to enable the notification on the Android side,   the error message returns Descriptor Not Found. The link that you attached is for iOS, maybe the Linkit 7697 works with iOS but not Android, you can use the app call "nRF Connect" from the Android store for debugging this issue. This tool app is uploaded by the nRF official. It has the same problem when reading the LBLECharacteristicBuffer, will get what ever size it is in a single packet.

Thanks for your great works, and best regards....

pablosun, 2018-01-09:
Hi, can you download the attached quick fixes:
[updated_LBLE_lib.zip](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/files/1614362/updated_LBLE_lib.zip)

and replace the unzipped content to `%USERPROFILE%\appdata\Local\Arduino15\packages\LinkIt\hardware\linkit_rtos\0.10.12\libraries\LBLE\src`? (Assuming you are using Windows version of Arduino IDE).

Hope this works.

weixiongmei, 2018-01-09:
Hi pablosun,

Very appreciated, the notification works on Android now. When there's a chance, hope you can add the MTU adjustment, txPower adjustment, and changeable characteristic attribute adjustment(read, write, notification, indication). These features are very useful when developing BLE projects.

Best regards..... Thank you~~~~

commit 482cb818 (2018-01-09): Fix #80 by handling CCCD for all characteristics

Some BLE implementation requires the presence of CCCD(Client Characteristic Configuration Descriptor) to propery "subscribe" to notifications and indications.

By default we allow notification/indication of all characteristics, so we add CCCD to all characteristics and corresponding read/write callbacks.

Note that this increases the callback table usage, since 1 characteristics now requires 2 callback functions. Therefore, the number of statically allocated callback table is doubled from 30 to 60 entires.
PR #84 (2018-01-10): BLE Update
This is a bug-fixing update for `LBLE` library.

Fixing Issues:
 * Fix #82 by fixing UUID comparison between 16-bit and 128-bit UUIDs 
 * Fix #80 by handling CCCD for all characteristics

Other minor updates:
 * Change type of txPower to signed integer
 * Adding Arduino IDE keywords
 * Fix build warnings
commit 47a16e20 (2018-01-10): BLE Update (#84)

Fix #80 by handling CCCD for all characteristics

Some BLE implementation requires the presence of CCCD(Client Characteristic Configuration Descriptor) to propery "subscribe" to notifications and indications.

By default we allow notification/indication of all characteristics, so we add CCCD to all characteristics and corresponding read/write callbacks.

Note that this increases the callback table usage, since 1 characteristics now requires 2 callback functions. Therefore, the number of statically allocated callback table is doubled from 30 to 60 entires.

* Fix #82 by fixing [... cut]
closed 2018-01-10 by pablosun