MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #83: UART baud rate 921600 not supported bug

weixiongmei, 2018-01-09:
## Reproduce Steps

## Error & Expected Results
the baud rate 921600 is not supported by default, because the bug of the code in the UARTClass.cpp, but this bug can be fixed by replacing the function "uart_translate_baudrate(const uint32_t baudrate)"  with the following code：

static hal_uart_baudrate_t uart_translate_baudrate(const uint32_t baudrate)
{
	switch(baudrate)
	{
		case 300:
			return HAL_UART_BAUDRATE_300;
		case 1200:
			return HAL_UART_BAUDRATE_1200;
		case 2400:
			return HAL_UART_BAUDRATE_2400;
		case 4800:
			return HAL_UART_BAUDRATE_4800;
		case 9600:
			return HAL_UART_BAUDRATE_9600;
		case 19200:
			return HAL_UART_BAUDRATE_19200;
		case 38400:
			return HAL_UART_BAUDRATE_38400;
		case 57600:
			return HAL_UART_BAUDRATE_57600;
		case 115200:
			return HAL_UART_BAUDRATE_115200;
		case 921600:
			return HAL_UART_BAUDRATE_921600;
		default:
			return HAL_UART_BAUDRATE_MAX;
	}

	return HAL_UART_BAUDRATE_MAX;
}

## Environment

 - Arduino BSP Version: 0.10.12
 - OS(Windows/macOS/Linux): Windows
 - Additional external peripherals:

pablosun, 2018-01-10:
Good suggestion & fix, will apply to master branch.

pablosun, 2018-01-15:
Added in [v0.10.13](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/releases/tag/v0.10.13)

closed 2018-01-15 by pablosun