MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #87: Can support BT-HID Device?

kyumdlab, 2018-01-15:
Both ESP32 and nRF8001/nR51822 have libraries that support BT-HID, 
Is Linkit-7697 planning to support BT-HID?

pablosun, 2018-01-15:
Thanks for asking! 

### Short answer
Not now, but we're open to suggestions on new functionalities. Do you want to use LinkIt 7697 as a keyboard/mouse and connect to your mobile phone? or desktop PC?

### Long answer

For clarity, by "BT-HID", do you mean "Classis HID profile" or "HOGP"?

For BT-HID, there are 2 cases:
 * BT **classic** HID profile: this is **not** supported, since LinkIt 7697 only supports Bluetooth Low Energy spec, not classic profiles.
 * HID-Over-GATT-Profile (**HOGP**): this is a new profile based on BLE (GATT). LinkIt 7697 can support this profile, since LinkIt 7697 supports GATT. However, it is up to the user to implement the HOGP profile and the bonding process.

For HOGP:
 * LinkIt 7697 + LinkIt SDK v4.6: There is already an example project that implements HOGP.
 * LinkIt 7697 + Arduino (This Github Repo): Currently not supported. Also, since a HOGP device requires bonding/pairing to work, we'll have to update the **LBLEPeripheral** library first to support bonding/pairing.

kyumdlab, 2018-01-15:
Thank you for your reply!

I want to use Linkit to connect to my PC / Mac.

I could use [arduino-BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral) to make my arduino a HID keyboard, and now I want to try Linkit-7697 instead.

Thank you!

pablosun, 2018-01-18:
The **arduino-BLEPeripheral** is basically HOGP-based. We'll investigate if it is feasible to either port it or provide similar functions with existing LBLE librarys for LinkIt 7697.