MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #89: Unreliable lookup method used within LBLEEventDispatcher class

sfyang, 2018-02-06:
In the [LBLEEventDispatcher](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/blob/master/middleware/third_party/arduino/hardware/arduino/mt7697/libraries/LBLE/src/LBLE.h#L232) class defined in [LBLE.h](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/blob/master/middleware/third_party/arduino/hardware/arduino/mt7697/libraries/LBLE/src/LBLE.h) and implemented in [LBLE.cpp](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/blob/master/middleware/third_party/arduino/hardware/arduino/mt7697/libraries/LBLE/src/LBLE.cpp), the [EventTable](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/blob/master/middleware/third_party/arduino/hardware/arduino/mt7697/libraries/LBLE/src/LBLE.h#L245) type is a [std::multimap](http://en.cppreference.com/w/cpp/container/multimap) container, and the implementation of both the `dispatch()` and the `removeObserver()` methods uses `std::multimap::find()` method to lookup the iterator to the first element in the `m_table` list.

However, according to the C++ standard, the `std::multimap::find()` method [does NOT guarantee to return the iterator of the first element for elements of the same key in the list](http://en.cppreference.com/w/cpp/container/multimap/find), so the outcome of the these two methods could be STL-implementation-dependent.

A possible workaround could be replacing the `std::multimap::find()` method calls with `std::multimap::lower_bound()`, which [returns an iterator pointing to the first element that is not less than the specified key.](http://en.cppreference.com/w/cpp/container/multimap/lower_bound)

pablosun, 2018-02-06:
Thanks for report this! We will investigate further into this issue.

pablosun, 2018-02-14:
Will use `equal_range` instead when searching for elements with matching keys.

commit 5f69fc70 (2018-02-06): Fix #89 Incorrect memset usage
closed 2018-02-26 by pablosun