MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #91: Some characteristics are missing if performing characteristic discovery with the LBLEClient class using LBLE library

sfyang, 2018-02-07:
This issue has been briefly mentioned and demonstrated in the issue report of #90.

Consider the following modified version of the 'ConnectPeripheral.ino' example from LBLE library, which inherits the `LBLEServiceInfo` struct and puts the characteristic list into the struct (thus probably solves #90 to some extend), and also extends the LBLEClient class to use the new `ABLEServiceInfo` struct:
```C++
/*
  This example scans nearby BLE peripherals and prints the peripherals found.

  created Mar 2017 by MediaTek Labs
*/

#include <LBLE.h>
#include <LBLECentral.h>
#include <iterator>

extern "C" {
#include <bt_gattc.h>
}

struct ABLEServiceInfo : LBLEServiceInfo
{
  std::map<LBLEUuid, uint16_t> m_characteristics;
};

class ABLEClient : public LBLEClient
{
public:
  bool connect(const LBLEAddress& address)
  {
    bool done = LBLEClient::connect(address);

    if (done) {
      discoverServices();
    }
  }

  int getCharacteristicCount(int index)
  {
    if (index < 0 || index >= getServiceCount()) {
      return 0;
    }

    return m_services[index].m_characteristics.size();
  }

  LBLEUuid getCharacteristicUuid(int svcIndex, int chrIndex)
  {
    if (svcIndex < 0 || svcIndex >= getServiceCount() || chrIndex < 0 || chrIndex >= getCharacteristicCount(svcIndex)) {
      return LBLEUuid();
    }
    auto c = m_services[svcIndex].m_characteristics.begin();
    std::advance(c, chrIndex);

    return c->first;
  }

protected:
  std::vector<ABLEServiceInfo> m_services;

  int discoverServices()
  {
    if (!connected()) {
      return 0;
    }

    bt_gattc_discover_primary_service_req_t searchRequest = {
      .opcode = BT_ATT_OPCODE_READ_BY_GROUP_TYPE_REQUEST,
      .starting_handle = 0x0001,
      .ending_handle = 0xFFFF,
      .type16 = BT_GATT_UUID16_PRIMARY_SERVICE
    };
[... 564 lines cut ...]
		 - e604e95da759481787d3aa005083a0d1
		 - 000000a50000100080000026bb765291
		 - 0000001d0000100080000026bb765291
		 - 0000001e0000100080000026bb765291
		 - 000000230000100080000026bb765291
	 - 000000440000100080000026bb765291
		 characteristics =
		 - e604e95da759481787d3aa005083a0d1
		 - 000000050000100080000026bb765291
		 - 0000001f0000100080000026bb765291
		 - 0000000e0000100080000026bb765291
		 - 000000370000100080000026bb765291
		 - 000000220000100080000026bb765291
		 - 000000010000100080000026bb765291
		 - 000000190000100080000026bb765291
		 - 0000001a0000100080000026bb765291
		 - 0000001c0000100080000026bb765291
Device fc:86:7a:4e:e5:f1(random) disconnected.
```

Comparing the output of these two cases, it is obvious that the scan result got from the LBLE library lacks lots of 128-bit UUID characteristics. For services containing more than 2 characteristics with 128-bit UUID, **at most two** of them are returned.

However, if further extending the `LBLEClient` class by adding methods for descriptor discovery, one can find that those missing characteristics, along with some attribute used by ATT, will be returned when waiting for `BT_GATTC_DISCOVER_CHARC_DESCRIPTOR` event after making a  `bt_gattc_discover_charc_descriptor()` call, which by itself is another erroneous behavior of the LBLE library.

It looks like something is not quite right in the FreeRTOS glue code used by LBLE library.

pablosun, 2018-02-09:
I can reproduce this locally. I'll check with the underlying module owners for further investigation. From the error code, it seems that `bt_gattc_discover_charc` reports error `0x3FFFFF6`(connection in use) and callbacks with status code `0x81` when querying for the 3rd argument.

pablosun, 2018-02-12:
Instead of calling `bt_gattc_discover_charc` again after receiving `BT_GATTC_DISCOVER_CHARC` with `BT_ATT_ERRCODE_CONTINUE`, we should simply wait for successive events and keep inserting them into the characteristic list.

The attached modified version of `ABLE` is a quick workaround that lists all the 128-bit characteristics of the chosen device.
[ABLE_multiple_event.zip](https://github.com/MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK/files/1715400/ABLE_multiple_event.zip)

sfyang, 2018-02-12:
I can confirm that, with the "Multiple" variant of both the `EventBlocker` class and the corresponding `waitAndProcessEvent()` helper, a properly modified and extended LBLEClient class with method for descriptor discovery can now work correctly for both characteristic and descriptor discovery.

closed 2018-02-26 by pablosun