MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #96: Potential Memory Leaks

pablosun, 2018-02-26:
https://zh.forum.labs.mediatek.com/t/7697-memory-leak/1382

2 memory leak cases are reported.

## Reproduce Steps

### LBLE
Heap memory reduces when calling this repeatedly
```
LBLEPeripheral.advertiseAsBeacon(beaconData);
LBLEPeripheral.stopAdvertise();
```

### Wi-Fi
Leaks when calling
```
WIFI.BEGIN(ssid, pass);
WiFi.disconnect();
```
repeatedly.

pablosun, 2018-02-26:
I'll try to reproduce these 2 issues.

chengc843988, 2018-02-27:
The BLE API, when calling advertiseAsBeacon, API clear old data with <vecotr>.clear(), then copy the new data into new object , 
so it could not free release the old *OBJECT* in the heap.
if have the API not to release old object, and uses the caller's object memory addreess, this issue is solved.

But,  I don't think this a good solution!!

pablosun, 2018-03-01:
@chengc843988 Thanks for the feedback, can you explain more on the `clear()` usage issue? 

There are several data members that get `clear()`-ed:

 1.  `std::vector<LBLEAdvDataItem> m_advDataList;`
 2. `LBLEAdvDataItem`
 3. `std::vector<bt_gatts_service_rec_t*> m_records;`

Item 1 and 2 seems pretty safe to be `clear()`-ed to me. We removes the element but keep the size of the vector, because we assume that the data size of advertisements are bounded by spec.

for item 3, it is true that the buffer pointed by`bt_gatts_service_rec_t*` never gets released. We assume that all services are declared statically and never gets released.

Perhaps you meant some other bugs? Can you clarify more? 

Many thanks for your help.

chengc843988, 2018-03-14:
LINE: 1066 (LBLEPeriphral.cpp)
// remove the advertisement data
// so that advertiseAgain() won't succeed
// until advertise() API is called again.
    LBLEAdvertisementData* ptr=m_pAdvData.release();
    delete ptr;