MediaTek-Labs/Arduino-Add-On-for-LinkIt-SDK #105: 7697D DA issue

medamarko, 2018-09-18:
Im getting upload stuck in waiting for DA response.
```
Start uploading the download agent
Waiting for DA output...
```
I must say im using D version of 7697 with dual band WiFI. Does the prebuilt da97.bin supporting this IC?

## Environment

 - AcSIP AI7697HD 
 - OS(Windows/macOS/Linux): Windows 10

Thank you!

pablosun, 2018-09-19:
Hi,

Unfortunately The prebuilt download agent (`da97.bin`) is built for MT7697 without compatibility to different models, e.g. MT7697D. 

You may want to contact the module vendor for the FlashTool program. 

I'll check if it is possible to use the same DA for both MT7697 and  MT7697D. There are other potential compatibility issues like Wi-Fi firmware and flash layout compatibility, so there is no guarantee that it would work.

medamarko, 2018-09-19:
Thanks,
i was able to get things running with help from AcSIP. FTDI IC latency param had to be lowered to 1. 
BLE portion of stack is working i will test now WiFI....

closed 2018-09-19 by medamarko